tesLAXApp/tesLAX #1: cell voltage histogram?

push2eject, 2020-03-17:
This app looks great, and I love the display configurability, but I'm trying to find an iOS app that can display more detailed BMS data. Specifically, I'd like to be able to see a histogram of all 96 cell-block voltages (and temperatures) as a visual indication of cell-balance. Is this possible?

tesLAXApp, 2020-03-17:
Hello.   The app doesn't have graphs yet.  No promises, but that would likely be a 2.0 feature, which is to say, quite a long way away (if ever)  There are a bunch of things on my list first, and I'm not quite sure how I'd integrate graphs yet.

I assume if you're interested in cell info, you're using a Model S/X.  There isn't really anyone maintaining a list of signals for the Model S/X like Josh does for the Model3, so the signals included in the app might not be as high quality as the Model 3 ones (I have a 3 myself, so I can't really test the S/X stuff, thats why I say "designed with Model 3 in mind, but may be useful for S/X")

How do you prefer to see cell balance?  Some sort of max/min/average/std-dev?  You could possibly accomplish something like that using javascript-based signals.  For example, the default (Free) view contains average cell voltage.  That signal isn't a real signal.  Instead it is a bit of javascript that runs every time a cell message comes in:

```
function pad(num, size) {
    var s = num+"";
    while (s.length < size) s = "0" + s;
    return s;
}

var total = 0.0
var count = 0.0
for (cell = 1; cell <= 96; cell++) {
  var variable = "CellVoltage"+pad(cell,2)+"_6F2"
	var thisCell = null
	try {
		thisCell = eval(variable)
	} catch(err) {}

	if (thisCell != null) {
		count += 1.0
		total += thisCell
	}
}
total/count
```

The cell voltage signals come in a few at a time.  This does a try to access each cell voltage as they come in over time, so staring out the average will be only a few voltages, and over time the average becomes more accurate.

push2eject, 2020-03-17:
I'm okay with it being biased to the Model 3 (I have a P3D, no S/X) :-), but I am still interested in cell balance. I normally only charge to 70%, and I was having a discussion with someone who told me I should charge to 100% regularly to maintain cell balance. I'm not keen on that, but I didn't have enough data to support my argument! I guess if a histogram isn't possible at the moment, then the main numbers I am interested in are the average and the peak variation (max-min). Std-dev also sounds like an interesting thing to monitor, as I guess it would put a number to how rough a histogram might look.

tesLAXApp, 2020-03-17:
Oh, sorry-- I misunderstood.  The Model 3 does not broadcast individual cell voltages on the CANbus.  My understanding is that many versions ago, it did broadcast the cell voltages, just like the S/X still does... but Tesla removed it for some reason.  My guess is that it wasn't being used and they needed the bandwidth for other messages.

I would join the discussion on [TeslaOwnersOnline](https://teslaownersonline.com/threads/diagnostic-port-and-data-access.7502/page-74#post-278371) to see what's might be possible on the Model 3.  I think there are max/min voltage signals in the database that might give you some insight.

Sadly for the Model 3, there just isn't nearly as much cell-level information being broadcast on the bus.

push2eject, 2020-03-17:
Oh - that's disappointing. I had seen the references to cell voltages in the Model-3 CAN spreadsheet some time ago, and didn't know that info is no longer available. Thanks for the clarification. I'll keep playing around with your app nonetheless - nice work! Cheers, Kevin.

jake-b, 2020-03-21:
I will close this issue, but graphs are on the long-term enhancement list.  Sadly cell voltages are not up to me-- if Tesla ever adds those back to the Model 3, then the app stands ready to decode them.

closed 2020-03-21 by jake-b