tesLAXApp/tesLAX #2: Paring required

jonasman, 2020-03-21:
Unfortunately it does not work with my canbus reader. My canbus reader works with TM-SPY app for ios.
This app requires odb pairing with BT, but the iPhone says it cannot pair for some reason.
Any change to not need pairing?

jake-b, 2020-03-21:
I am not sure which CANBus reader you are using, but currently the only supported accessory is the "OBDLink MX+" (as noted in the app listing)

There are some Bluetooth Low Energy (BLE) CANBus readers out there that work with other iOS apps, but they are not currently supported by the application.  I have not found a BLE adapter with an STN1100 chipset, have found to have better performance/filtering.

Right now  ELM 327/BLE adapters are not my top priority, unfortunately.   As this is a weekend/side project I have to pick my battles.

For reference, which CANbus product do you use?

jonasman, 2020-03-21:
I bought from this german site: https://e-mobility-driving-solutions.com/produkt/ciagnostic-adapter-cable-facelift-models-modelx-bundle/?lang=en

"ELM327"

jake-b, 2020-03-21:
Yes, the ELM327 is a different chipset.  For reference, here is a comparison between the ELM327 and the STN1100

If you're curious to see the difference, here is a comparsion: https://www.scantool.net/scantool/downloads/102/stn1110_vs_elm327.pdf

Interesting also is that the product page does mention varying performance quality of the cheap ELM327 dongles and links to the OBDLink MX+ amazon page.

I'm not saying "never" for ELM327 support, but as I said, many more priorities. :)

jonasman, 2020-03-21:
I totally understand your point!
The site says the ELM is cheaper and works with TM-SPY. 
I tried to pair the phone with the ELM adapter but it gives an error saying pairing not supported. Even thought the adapter come with a pairing pin. 

I guess I will continue to use the TM-SPY app until one day..

jake-b, 2020-03-21:
I will close this issue but like I said, this is on my enhancement list.

closed 2020-03-21 by jake-b