tesLAXApp/tesLAX #3: App crashes when trying to add a new signal

ChadMoran, 2020-03-30:
When trying to add any kind of new signal to the main screen the app crashes.

jake-b, 2020-03-30:
Thanks for the bug report.  That definitely shouldn't behave that way.

There are some lurking crashes in those editor screens.  I'm suing Apple's new SwiftUI framework and it tends to fail in unexpected and creative ways.  However, if this is a consistent reproducible crash, then its probably my issue.

What signal are you trying to add and in with what options?  Can you send a screenshot just before the crash?  I'll try to replicate.

There is one big hammer you can try to resolve the issue-- if the database was corrupted somehow, you can try to reset the app's database.  This will reset everything back to default (including any custom signals or gauge configurations).  To do this, you go to the default iOS settings app and find the tesLAX page, and check the box to reset the database.

I'll look to see if Apple sent me any crash logs that can help me narrow down the problem.

ChadMoran, 2020-03-30:
I just click the button in the upper-right and select `Add New...` > `Static Text` and it crashes reliably 100% of the time.  Doesn't matter if I am or not connected to a BT device.

ChadMoran, 2020-03-30:
iPhone 11 Pro Max.  I tired on an iPhone X Max I had laying around and it worked fine.

jake-b, 2020-03-31:
This is interesting to me as there shouldn't be a difference between the X Max and the 11 Pro Max.   I would try the database reset on the misbehaving device, and see if that clears up the issue.  Its possible the DB is corrupted somehow on the one that doesn't work.

Also, I'm adding console logging to the next version of the app so that it'll be easier to collect logs of issues like this to troubleshoot.

ChadMoran, 2020-03-31:
Reset the database.  Rebooted the device.  Reinstalled the app.  No dice.

ChadMoran, 2020-03-31:
I also found opening certain screens like "Presets" will randomly crash the app.

jake-b, 2020-03-31:
Hi there.  I don't have a X or Xs to test with, but I did replicate the crash (I think) in the simulator.  I'm not sure why it occurs.  Something to do with Swift Generics and SwiftUI  I think.  It should be fixed in the next version.

The Presets screen (and random other screens) are also obscure SwiftUI crashes and are fully on Apple's side.  Luckily it looks like the new 13.4 and SDK may have fixed some of that instability.

jake-b, 2020-03-31:
Also, if you want to join the test flight, send me an email with your email address and I can add you to the beta when I get around to another beta build.

ChadMoran, 2020-03-31:
I sent a crash report through TestFlight if that helps.

ChadMoran, 2020-04-02:
Latest TestFlight build fixes both crashes I was seeing.

jake-b, 2020-04-03:
Thank you. I will close the issue.  Still more work to do on that build before it's ready for the AppStore, but hopefully the TestFlight will get you by until then.

closed 2020-04-03 by jake-b