tesLAXApp/tesLAX #5: Preset editing

bwilson4web, 2020-05-15:
I have the extended subscriptions and can copy a Preset and add to that preset. However deleting a gauge or changing the display does NOT save in the preset. I have to enter all changes each time instead of having a saved, named Preset.

jake-b, 2020-05-15:
Hi there.

Thanks for the bug report.  This should not behave like this.  You report two issues:

1. Removing a gauges does not remove it from the preset.  I see this behavior and will work on the fix.

2. "changing the display" does not save.  Can you speak more to this?  Are you referring to the number of columns?  Right now the number of columns is global, not a per-preset thing.

Can you elaborate specifically what's not saved?  I am able to add new gauges and reorder them and they get saved.  If you are not seeing this behavior, let me know and I'll look more closely.

bwilson4web, 2020-05-15:
Hi,

I made a short YouTube showing the problem but it did not include the
problem with different Presets. I'll make one later this afternoon.

https://youtu.be/C_ET7ma1dmE

As a suggestion, some sort of "save" button on a Preset would help ... or
perhaps I missed it.

Bob Wilson

jake-b, 2020-05-15:
There is no save button because it is supposed to keep things saved.

With deletes, the bug is that it it isn't saving the delete to disk, so when it reloads the preset, it reloads things from disk that should have been deleted but weren't.

Other operations, such as create, reorder, and modify should get committed to disk properly.

Also, if you want to join the test flight, send me an email with your email address and I can add you to the beta.

jake-b, 2020-05-15:
Thanks I can reproduce the 2nd bug too.  I'll take a look at it.

bwilson4web, 2020-05-15:
"Also, if you want to join the test flight, send me an email with your
email address and I can add you to the beta."

I would be happy to join the testing team: bwilson4web@gmail.com

BTW, I've attached some screen shots showing:

tesLAX_Presets -1.jpeg - restored to original configuration  ... excellent
feature
tesLAX_Presets - 2.jpeg - all of the fields of the restored configuration
... no problem
tesLAX_Presets - 3.jpeg - the copy did not include the "restore" and fields
options

Notice I tried to copy the empty "Other CANBus Device" but had no gauge
library to select signals. I thought maybe I could just fill out a "blank"
but without signals ... no luck.

Bob Wilson

jake-b, 2020-05-20:
Please let me know if the 1.1.1 beta is better behaved.  Thanks!

bwilson4web, 2020-05-20:
Will do!

Tuesday was busy but Wednesday looks good.

Bob Wilson

jake-b, 2020-12-23:
I believe this issue has been resolved in recent versions.  Please re-open if you still have issues.

closed 2020-12-23 by jake-b