tesLAXApp/tesLAX #6: Problem importing DBC file

pk-r7, 2020-06-02:
I tried to import DBC files created with both Kvaser and Vehicle spy. The app just gives the spinning motion and never imports the file. I cannot see any error logs otherwise I would supply them. If the creator wants to reach out to me directly I can supply the DBC files.

jake-b, 2020-06-02:
Hi.  If you're willing to share, please send the DBC to support (at) allaboutjake [dot} com.  I won't use or share it beyond debugging the import.  Also, if you want to join the TestFlight beta to test any bug fixes, indicate so in your email and I'll add you to the TestFlight.

Thanks for the bug report.

jake-b, 2020-12-23:
Hi there.  Does this issue still occur in recent versions fo tesLAX?