tesLAXApp/tesLAX #7: couple of feedback points

nebukadnezar, 2020-07-16:
Love the app, subscribed annually to help support.

Some ideas:
- Create an additional gauge type that shows the minimum/maximum values in one visual display. This would be very useful for e.g. cell voltages or temperatures. Look at the arcInstrumentDual display in the instrumentation library I wrote many years ago (bottom left dial): https://www.inside.net/ISinstrLib/  - this is similar in concept to what cockpit displays in airbus type aircraft look like.

- add some form of timestamping feature for the logged CAN data. This would make it easier to locate specific events, especially when trying to decode new frame types.

- and lastly, I'd like to create aggregate dials using values from several points to calculate new values. For instance, I'd like to show the total power consumed at a point in time which is simply the HV battery voltage * HV battery amps. Perhaps that's already possible and I can't see how to do it?

It's a great app, and I love the fact it can play back recorded files. Very neat!

jake-b, 2020-07-16:
Hi there.  Thanks for the great feedback.

I am actually in the middle of rewriting the gauges.  Currently they support 3 types of ranges:

   1. Static - meaning the max/min of the gauge is what's defined in the signal database. This isn't so useful because some signal ranges are large but in practice you don't see them.  For example, battery voltage can be 0v - 655v.  By signal definition, it is 11 bits scaled at 0.01 -- so 2^16*0.01 being the max value.  In practice, however, you really only see 300-400v unless something is really wrong....
   2. Lifetime Min/Max - basically the app keeps track of values it sees and will set the top/bottom of the scale to this value. (currently a bug prevents this from being saved between app launches, oops... fixed in the next version)
   3. Drive Min Mix - same as lifetime, but resets every time you start a new session.

I guess I could do something similar to your ask with the static mode, show the lifetime or min/max as an additional arc.  I'll consider it for the future...

I my next beta, the new gages are resizable and render faster, so stay tuned.

Regarding timestamping, I might be able to do something like that for the ASC file format logging.   I have to see if the file format has provisions for such a thing.  I'll look into it.

The app does support "Calculated Signals" -- there is some info in the help file but it is not a very well used feature and hence hasn't really been put through its paces to see how robust it is.

Basically, you can create a signal and attach it to a message.  Set its type to "calculated".   Write some JavaScript in the 'Script' field.  Attach a gauge to this signal in your layout.  Then any time that message comes in from the CANBus tesLAX will execute the javascript and render the result in the gauge. 

Some notes from the help on this:

    - All signals in the message are decoded and set as variables within the message.  This can be used to create calculate signals from signals in the same message.
   - All the bytes of the message are assigned as an array called message[] in the JavaScript context.  This can be used to implement custom decoding of the bits.
   - Signals that are not in the "trigger message"  may also be available if they've been seen previously.  Keep in mind, that if you don't have gauge in your layout for a given signal or message, then tesLAX may have a filter set and it won't be "seen" by the app.  Effectively, you can use calculate based on  signals in the "triggering message" and any messages you are currently visualizing elsewhere.   Note also that the data in the "triggering message" is current (because it is the message the triggered the calculation and the data in that message is fresh) but any signals from other messages may be older depending on when it was last "seen" on the canbus.

Its hard to write and debug these, so they're best kept simple.

I know its confusing, so hopefully that makes sense.  Search the Model 3 preset for "BattPower" or the Model S/X presets for "Average Cell Temp" for some examples of this in use. The "AverageCellTemp" one is a good example because it also uses a Muxed signal, meaning each second or so a new message comes in with some new cell temps/voltages.  Over time, the app eventually sees all 96 values, but during the process, the average starts out with 1 one value, then 2... then 3... and eventually averages all 96.  (This signal isn't present on the Model 3 sadly, which is why its only in the S/X preset)

Hope that helps.  Feel free to send any more ideas or questions.

nebukadnezar, 2020-07-16:
Thanks for your detailed response! I have a Model S performance and have created a voltage differential gauge using the code from your average cell voltage code, but modified to return the difference between the lowest and highest cell. So that works fine. I'll need to look into your other points.

There are several parameters that aren't working with the S correctly. The Battery amps for instance are scaled differently for the S than they are for the 3 (assuming it returns sensible values for you in the 3 - I can't test that).

One parameter I am very interested in is finding out what autopilot modes are engaged: TACC, TACC+AS, and NoA. I think message IDs 0x256 have some cruise control parameters, I'm about to test that today by recording a short snippet (let me know if you want a copy and I can post it). This again may or may not be different for the 3.

In regards to the DBC files, what reference are you using for your implementation?

The gauge type I was proposing wasn't meant to reflect min/max values encountered during lifetime/drive as your current gauges do, but those are meant to reduce a set of many values to show min/max and avg - for example the cell temps or cell voltages. So they report the current instantaneous range of values, not a range based on historical data. For that gauge you'd set the min value of the gauge to the lowest battery voltage currently reported, the max value to the highest, and the average or median would be in the middle.

jake-b, 2020-07-17:
I understand the what your asking for now.  I'll put it on the list as a possibility for the future.  For now, the work around would be a few javascript-based gauges as you seem to have figured out.

Regarding the DBC for the S, unfortunately, the data in the app is probably just wrong or out of date.   I don't have a Model S/X of my own and there is no one maintaining a current DBC for the Model S that I know of.  Most of the data in the app was pieced together from [one person who posted a reverse engineering of the signals a long time ago](https://skie.net/uploads/TeslaCAN/Tesla%20Model%20S%20CAN%20Deciphering%20-%20v0.1%20-%20by%20wk057.pdf), forum posts and random sources.  Its probably not up-to-date. 

If you have any updates to the database, or have a better source DBC that can be published in the app, let me now.  (Same with that battery difference signal, if you want to contribute that to the default database, I'm happy to include it)

One note, the you might want to save any of your javascript code in a note or text file somewhere just so you have it.  While I do my best to keep the database compatible across updates, sometimes it can't be helped.   I eventually want to have a "preset export" feature on the app so you can backup and share presets (Its on the list!)

Also, if you want to join the TestFlight betas, find my email address or send me a private email with your email address and I'll include you in the next TestFlight.

Regarding AP signals, I know the Model S/X debug connector has 3 canbuses in it.  I'm sure that data is available somewhere, but it might not be on the Chassis bus. I'm not sure which bus is the default bus that people connect their MX+ too on the S/X.  I've been experimenting with the Comma.ai Panda, which supports 3 busses simultaneously, but support for that dongle is very early and really only for advanced users and requires an adapter or non-standard USB cable to get working for the first time.

Thanks!

nebukadnezar, 2020-07-17:
Ok, I'm happy to contribute what I know/found. Is there a way to backup/restore the presets? AllI can seem to find is how to duplicate them. It also would be very useful if able to edit presets and even gauges on the computer and then import them... it's a bit tedious to type it all on the iPhone.

jake-b, 2020-07-18:
Sadly, not currently.  I would like to add the ability to backup, share, and restore presets eventually.  I agree it is a bit tedious on the phone.

nebukadnezar, 2020-07-23:
Can you please look into making it possible to access all signals being decoded from the javascript code of a gauge?

nebukadnezar, 2020-08-07:
Forgot to report back that I figured out how to do this - all variable names are accessible from any script, that's really neat! Is it possible to use persistence between the scripts being called? I'd like to dampen some of the values by e.g. calculating a running average of the last 10 values and using that for the gauges. Also, I emailed via support@ allabaoutjake a couple of weeks ago for test flighting, but never heard back.

jake-b, 2020-08-07:
Sorry about that.  I missed your email and I have added you now.

You should be able to create variables as you say.  Take a look at the Model S/X presets for how the average cell voltages are calculated.

```
function pad(num, size) {
    var s = num+"";
    while (s.length < size) s = "0" + s;
    return s;
}

var total = 0.0
var count = 0.0
for (cell = 1; cell <= 32; cell++) {
  var variable = "CellTemp"+pad(cell,2)+"_6F2"
	var thisCell = null
	try {
		thisCell = eval(variable)
	} catch(err) {}

	if (thisCell != null) {
		count += 1.0
		total += thisCell
	}
}
total/count
```

jake-b, 2020-08-07:
I misread.  I think you should be able to create global variables any they will persist across script executions.  I never tried.

jake-b, 2020-12-23:
I think we're good with this one :)

closed 2020-12-23 by jake-b