tesLAXApp/tesLAX #8: UI enhancement: Smaller gauge layout

dburkland, 2020-07-24:
I have been a long time user of Scan My Tesla however prefer iOS and thus trying to migrate my custom Scan My Tesla dashboards over to tesLAX. With that said would it be possible to add the ability to select a smaller gauge size so you could fit a few more rows of gauges on a given screen? I posted a reference shot of one of my Scan my Tesla dashboards along with one I am currently building out in tesLAX. 

[Scan My Tesla Dashboard](https://www.dropbox.com/s/5rhqjvan9mqm6q9/IMG_1280.jpeg?dl=0)
[tesLAX Dashbaord](https://www.dropbox.com/s/rvmfsp212pq80i5/IMG_0071.jpeg?dl=0)

Thanks,

Dan

jake-b, 2020-07-24:
Hi Dan,

Are you interested in joining the beta?  If so, just find a way to privately send me your email address and I'll add you to the TestFlight for the next beta.  I think you'll like some of the upcoming features, which address your feature request.

dburkland, 2020-07-24:
Absolutely! Are you on twitter? I can shoot you my address in a DM if that works.

Thanks,

Dan

jake-b, 2020-07-24:
That works. @AllAboutJake

jake-b, 2020-12-23:
I believe this issue has been resolved in recent versions.  Please re-open if you still have issues. :)

closed 2020-12-23 by jake-b