tesLAXApp/tesLAX #9: App shows N/A on all charts

nrm94ee330, 2020-08-03:
Hello,
I have OBDLink MX + OBD2 scanner  (https://www.amazon.com/gp/product/B07JFRFJG6/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)
After plugging in the scanner and when click on connect, it finds the bluetooth device. It says Starting connection for a little bit and then comes back and says "No Responce from Accessory" . None of the values on the chart show anything .
The default app does connect to the OBDLink MX+ and can get the basic data of the car. So I know the OBD Link works
and it also works on other ICE car as well.. so the OBD Link MX+ works ... I have a Model S 90D from early 2017.
Could you please help to figure out how to see data in the app from the scanner. 
Thanks

jake-b, 2020-08-03:
Hi there.

Here are some initial thoughts.   First, make sure that the device is connected in the Apple default Settings.app under Bluetooth first.  Also make sure that there are no other apps actively using the device.  It is also possible that a previous app left the MX+ in a unknown state so I also recommend power cycling the MX+ by unplugging and re-plugging the adapter, connecting and Setting.app, and then trying tesLAX first.

Some notes about Model S support.  The signal database is a little out of date for Model S.  There isn't anyone maintaining a nice, curated, up-to-date database of Model S signals.   I believe that SOC and BatteryCurrent signals may show false data.  I am investigating and hope to release an update with better signals for the Model S/X.

If you'd like to join our chat room for tesLAX, please visit https://discord.gg/DWpvvV -- I usually check in from time to time.

Jake

jake-b, 2020-08-05:
Hello again.  I just wanted to check in and see if you  have had any luck with this.

Also, I'm preparing a beta with some updated signals for the Model S.  I do not have a Model S or X to test against personally, and we don't have many beta testers with an S or X.  If you're interested in testing the latest beta, with these (hopefully) improvements, the send me an email or private message with your email address and I'll add you to TestFlight.

Also, you can find us on discord as well. https://discord.gg/cH9GwK

nrm94ee330, 2020-08-05:
Hello,
Thanks for following up..
I was able to figure out that the cable I had was/is faulty. I have ordered another one and waiting on it to arrive to test again.

Yes, I would definitely be interested in beta testing.
I will send you a PM shortly with my email address
Thank you

jake-b, 2020-12-23:
I believe this issue has been resolved in recent versions.  Please re-open if you still have issues.

closed 2020-12-23 by jake-b