tesLAXApp/tesLAX #10: Typos on the TESlax website

israndy, 2020-10-02:
Hope this finds you well. I was reading thru the teslax.app site trying to figure out how to customize my display. Discovered quite a few simple typos that I thought I would point out so they could be fixed.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

jake-b, 2020-10-02:
Thanks!  Yes, I should proof read things better, and documentation is often an afterthought.  (These typos likely exist in the help document in the app as well.)   I'll fix them up this weekend.  Thanks!

jake-b, 2020-10-03:
Typos on website (and in app help file) have been corrected.  Thanks for your attention to detail, as I'm usually more focused on the code then on the documentation. :)

closed 2020-10-03 by jake-b