tesLAXApp/tesLAX #12: Ipad gauge limet

LeoZalewski, 2020-10-21:
Great app
On iPhone "+ Add New" seams to be limits less
On Ipad "+ Add New" limits to about 16 gauges, related to where the "+ Add New" is on the screen.

jake-b, 2020-10-21:
Hi there.  Yeah, I've seen that before, and its a quirk of SwiftUI.  An upcoming update should fix this issue.

LeoZalewski, 2020-10-21:
Is it possible to rename a gauge description?

jake-b, 2020-10-21:
In the signal list, you can assign a "display name" to a signal which will change what is shown under the gauge.

Also feel free to join us on discord, to chat amongst other users: https://discord.gg/umZ8wvG

jake-b, 2020-12-23:
I believe this issue is resolved in recent versions.  If not, please re-open the ticket :)

closed 2020-12-23 by jake-b