tesLAXApp/tesLAX #13: iPhone 12 Pro plus gadge limit

maltalkstesla, 2020-12-22:
Hi I am a youtube blogger and I am featuring your product in a series I’m doing on my channel Mal Talks Tesla 
But have noticed when I use my iPhone 10x max the gauges are fully displayed On my iPhone 12 plus pro max it doesn't display fully can this be fixed also can you give me a list of the abbreviation  terms used  such as socu 1292
BMS these will be helpful when explaining last but not least one you set a new page how a you editors name as I don't see how this is done 
Regards
Mal

jake-b, 2020-12-23:
Hi can you send me a screenshot of the behavior you are seeing?

The signal and message names are inscrutable.  They are reverse engineered by the Tesla community as they are not provided by Tesla.  "SOC" for example, means "State of Charge".  BMS means "Battery Management System".  I don't know of anyone maintaining a full list of definitions. (And I'm not sure if we even know what they all mean)

Also, if you like you can join us for discussion on Discord at https://discord.gg/D6nKsCxegr - there is a small community where I and others can help answer your questions.

maltalkstesla, 2020-12-23:
Here is the picture [cid:B3C60ED9-3B32-42DC-B81A-BD1249EA9A44-L0-001]

[cid:17144051-AC61-440E-BA12-EEC4E062B0F0-L0-001]
As you see iPhone 12 pro max. Limits the view
Regards
Mal
If you can’t  Malcan

maltalkstesla, 2020-12-23:
Here is the view from I phoned X pro max

[cid:010A2750-2795-4242-833D-270A29D04A38]

[cid:9078BE2A-3689-48CD-B916-416B3F6E2C29-L0-001]
If you can’t  Malcan

On 23 Dec 2020, at 15:56, Mal Wayne <mal@malcan.co.uk> wrote:

 Here is the picture
<image0.jpeg>

<image1.jpeg>

As you see iPhone 12 pro max. Limits the view
Regards
Mal
If you can’t  Malcan

jake-b, 2020-12-23:
Looks like the GitHub email system ate the attachments.  Can you try posting through the website?

maltalkstesla, 2020-12-23:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

maltalkstesla, 2020-12-23:
Hopefully you got it. excuse other pictures there what I'm working on my YouTube channel 
Mal Talks Tesla,I'm using Teslax  to show what happens when the screen goes out

closed 2020-12-23 by maltalkstesla

jake-b, 2020-12-23:
OK, so if I understand correctly, I think you can make the layout a bit better.

Press the "(...)" button in the upper right, you can configure the size of the gauges (with the slider) and the number of columns.

Text truncation (when text get cutoff with "..." is controlled by the operating system's layout engine, so I don't have much control over that, unfortunately.  Worst case, you can edit the signals in the "Signal List" to provide a shorter, more descriptive 'Display Name' which will be used for the gagues.