tesLAXApp/tesLAX #14: Convert CSV files limited?

teddy-fuka, 2021-03-20:
I am happy to use this great application.
Especially, I can review the vehicle condition and reaction with log files of long distance trip after driving.
I am trying to create the graphs on Microsoft excel by using the converted CSV files.
At this time, I have two issues about the converting. 

First one is the limitation of the devices.
At first, I set the following 3 devices.
1: iPhone 8 (subscribed annually)
2: iPhone 12 pro (Extra Strength 1.0)
3: iPad Wifi model (Extra Strength 1.0)
Only iPhone 8 can convert to CSV, then I changed the purchasing to subscribed annually for iPhone 12 and iPad.
After that, iPhone12 can convert, but iPad cannot. Setting condition seems no deference.
Do you have any advice for iPad setting etc?

Second one is the limitation of recording time.
I can convert to the CSV files from log files with 5 minuets datum, but cannot from log files with longer than 10 minuets datum. 
I guess some file size limitation existing, is it helpful to eliminate the setting signals that I don't need?
I want to deal at least 1 hour log due to the driving range, If signal elimination is helpful, I will try next time.

Thank you for your advice in advance.

jake-b, 2021-03-28:
Sorry I missed this issue. Do you think this could be related to the iCloud issue we have found?

There should be no limit to length but the CSV convert is very slow. It’s on my list to see if I can improve. 

Also side note: check out [SaavyCAN](https://github.com/pyjamasam/SavvyCAN/actions/runs/546436375) (link is to recent build of a fork I use) it is an open source CAN file viewer that also does graphing.  It’s not super easy to use though. You first need to load in a DBC file that explains how to decode the data. Then you can load the log file and decode the data.  The best DBC for model 3 is still @joshwardell’s at https://github.com/joshwardell/model3dbc but may be a bit out of date.  Model S&X ones exist on the internet but I don’t have a good recommendation for one of those.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

teddy-fuka, 2021-03-30:
Hi jake-b

Thank you for your effort of improvement this application and important information.

I just find the reason of non converting to CSV files.
Some log files have a wrong header of them.
The following header have some mistake for example.

date 2021-03-28 22:06:57 +0000
date 2021-03-28 22:06:57 +0000
base hex  timestamps absolute
internal events logged
// version 11.0.0
base hex  timestamps absolute
internal events logged
// version 11.0.0
   -0.0000 1   381        Rx D 8 05  00  00  00  00  00  00  00
    0.0000 1   241        Rx D 7 3B  78  EC  0E  1E  5B  00
    0.0001 1   132        Rx D 8 4E  91  C1  FF  D7  3F  FF  0F
    0.0003 1   2C1        Rx D 8 07  00  00  00  00  00  00  00
    0.0016 1   263        Rx D 8 01  00  FA  3A  91  FA  D5  03

Repeating information of datum and also -0.0000 timestamp data brock the converting process.
Another log file save the data before saving the header.
The iPad usually make a wrong files. The probability of making it is much more than iPhones.
Please save the row datum after finishing to make the header.
These files cannot start to replay the log file unless sliding the time bar.
If possible, please improve this issue.

See you.

jake-b, 2021-03-30:
Thanks for this excellent bug report. I’ll investigate.  Not sure why the app generates malformed headers yet. I’ll look into it!

If you would like to join a group of users to discuss directly you can find us on Discord here:

https://discord.gg/P7tvgG27

Also if you’d like to join the beta, then I will need your email address to add you to TestFlight. Don’t post it here, send via email or private message on discord.

I’d you can send me a malformed log file that would be helpful as well.

Thanks!
Jake

teddy-fuka, 2021-04-01:
Hi jake-b

The test flight build completed some logs without header problems on the iCloud folder during these two days.
Also converting issue has no found. It seems be improved successfully.
Converting time is improved to half by previous build, but I spend  a lot of times.
I feel this is iPhone or i-OS performance issues. It's no problem. 

 Thanks a lot for your effort.

jake-b, 2021-04-01:
Thanks. I'll close this issue for now.  The performance is slow because there is no way to know ahead of time how many columns you need in the CSV file.  Each time a message comes in, it could have more signals and columns need to be added. 

The app currently renders this into a database and then writes the database out to disk at the end.  I probably could get more clever in how this works.

Side note, the ~2000 column limit for CSV export is because the database engine can't support more than that.  CSV files can get very big very quickly.  Its actually a huge amount of data!

Thanks again.  Reach out if you have any more questions.  This fix won't be in the App Store for a while so please keep using the betas as they appear and let me know if you have any other feedback.

closed 2021-04-01 by jake-b