tesLAXApp/tesLAX #15: Unable to find Logs

stefannilsson, 2021-03-20:
Got the XS features and a yearly subscription purchased, and logging has been turned on for some drives.
But I'm unable to locate the logs (for further download to computer).
Cannot find any hints of _where_ in iCloud Drive these files should be stored?

Apart from this, thanks for a kiss-ass app!

teddy-fuka, 2021-03-20:
Use the Files app or tap "open File..." in tes-Lax 
https://support.apple.com/en-us/HT206481
Logging files are located in the Tes-Lax folder in "On My iPhone"
Copy or move the suitable files to iCloud Drive.
If you cannot find them, another issue.

jake-b, 2021-03-20:
Hey thanks for the assist!

There is actually a bug in the most recent version of tesLAX that prevents it from writing to iCloud Drive.  I've sent a new build to TestFlight for approval by Apple, and will follow to the AppStore.  Probably going to take a week given all the approval steps.

In the mean time, the files are saved to the "On my Device" folder.

teddy-fuka, 2021-03-28:
I recognize that the new build tes-Lax saves log files to iCloud drive.
But the the log file on the iCloud drive cannot be converted to the CSV file.
Probably CSV files are unable to be saved to iCloud folder.
(the previous process was run but saving process cannot....)
I can convert the log file to CSV after relocating it to "On my Device" folder.
Could you please to add the option of selecting the folder to save the log file due to avoiding the relocation.
If there is already existing  such a function, please tell me the way.

jake-b, 2021-03-28:
Thanks @teddy-fuka — sounds like another iCloud related bug but let me investigate to find out for certain. Thanks for the report.

jake-b, 2021-03-28:
@teddy-fuka I have identified the issue. I’m not sure when the next release will be, but it should fix this issue.  

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

jake-b, 2022-10-06:
tesLAX 2.1 is rolling out now to the AppStore.  Its a staged roll out, so it might take a few days for it to hit your update queue.  Let men now if you still see this issue.  Thanks.

closed 2022-10-06 by jake-b