tesLAXApp/tesLAX #18: 12V battery gauges display vary so quickly that they’re impossible to read

Teslaric, 2021-09-27:
I have the tes-LAX app on an iPad and set up a 12v battery page for the basic 12v battery readings:  voltage, current (amps), capacity (AH),  and temp.  When the app is connected the readings vary so quickly that it’s impossible to read them.  Is there a solution to make the readings more stable?

jake-b, 2021-09-27:
Hi there.  Tesla has been changing those signals around in recent versions so it's hard to keep up.

Which vehicle are you on M3/Y/Plaid or S&X?  I'm assuming the former.

I'd make sure to have the 2021.12 preset selected, because there were some 12v signal updates in that version (over the 2021.4 version). If you still have things jumping around then its possible Tesla has made some more changes that break our decoding rules.

Also I think 12v battery temp has been broken for a while.   The community hasn't figured some of these things out yet.

Find us on discord if you want to chat. (https://discord.gg/kZxsgTqv) JWardell might know more about these signals than I do.

To answer your question though, "Extra Strength" users can edit the signal database to make it work, but you have to know what the decoding rules are, and unfortunately, Tesla changes things and doesn't publish the values.

Teslaric, 2021-09-27:
Thanks for the quick response.  Yes, I have a Model 3.  I experimented with placing the appropriate signal in the multiplexing signal, which yields a fixed reading.  I’m experimenting with charging the battery to see if the readings change.

jake-b, 2022-10-06:
tesLAX 2.1 is rolling out now to the AppStore.  Its a staged roll out, so it might take a few days for it to hit your update queue.  Let men now if you still see this issue.

closed 2022-10-06 by jake-b