tesLAXApp/tesLAX #19: Connection issue

angellee200, 2021-10-05:
Downloaded TesLAX demo versionon  iOS 15 (for iphone X )  ( full access in 5 days) , Purchased OBDLink MX+ and OBDLink CX, plug it in to two different car, 2016 Model X OBDII port and 2012 BMW E series, with the OBDLINK that came with the dongle , i managed to read the VIN off of my Tesla and the BMW, however when I tried to connect through TesLAX apps, it will start the connection then it time out and then disconnected, then it will loop the connecting and never get connect, yes I aware of the blue plug behind the center console , but if the OBDlink software manage to read the Vin off my Tesla, then tried it on my BMW , same thing happen? what exactly am I missing? I have been using the same OBDII with a CAN sniffing tools with success, can anyone shred some light what exactly went wrong ?

jake-b, 2021-10-05:
The OBDLink MX+ and the OBDLink CX have different connection methods. The MX+ is "Bluetooth Classic" and the CX is "Bluetooth Low Energy"

First, please make sure you granted the request for tesLAX to use bluetooth.  You can check this in the stock settings app under "tesLAX":

[image]

It might be worth force quitting tesLAX before starting just to make sure you're in a clean state.  If you're connected to a Tesla Model X, select the "3 & X EVs: Pre 2021 Refresh" preset to configure the app for this type of vehicle (unless you in a Plaid-refresh Model X, but I'm not sure there are any of those out there yet)

For the OBDLink MX+ you must be paired/connected to the device in the settings app before running tesLAX. Do this in the stock setting app and run tesLAX.

Then in the menu, select "Accessories" and "Bluetooth".  you should see the MX+ appear in the list.  Tapping should start the connection.

For the CX+ there isn't a pairing processing the stock iOS settings app, but there is the above mentioned bluetooth permission that must be granted. Again you might want to force quit the app first just to be sure you're starting fresh. It should appear in the same list that the MX+ does.  It also may ask you to "pair" the first time you use it.

Both devices take. few seconds to configure before data should start streaming. 

You can find me during most USA Eastern Timezone early evening hours on Discord along with a few other knowledgeable users and fans of tesLAX.  I'm happy to help you further if you prefer chat: https://discord.gg/weTnej4a

angellee200, 2021-10-05:
I attempted the force quit method and even restart my iphone, then uninstall and install the apps again with no success.   I did connect them manually in setting, I basically try everything above you have mention. 

For the gas car, I do have the DBA CAN signal imported, for BMW RPM RDL is in  0xAA , i tried to connect it in my BMW still the same result. 

I know the port has a CAN build in as I probe my Oscilloscope(decoder build in) with success , yes Can I set a appointment with you so I can try it while you are online? thank you.

angellee200, 2021-10-06:
I have Tesla Model X , I tried it with the OBDII port without success , then I realize the actual CAN access is at the blue plug ( I just order the connector) .  at the mean time, I figure it wouldn't hurt if I try it on my BMW, since I used Savvy CAN with success, by using listen only without any PID request (0x7E8 PID), it is streaming live... I will try it again when the connector come in. THank you thou.

jake-b, 2021-10-06:
Ahh yeah.  Some of the OBD-2 ports in Telsas are "just for show". Power only and no data pins connected.

You could probably receive that 0x7E8 broadcast if you knew how to decode it.   The easiest way would be to disable filtering (Settings->Log Recording Options->Disable Filter while logging).  This disables the filtering I spoke of earlier when recording a log file.  (Also you might want to change the log format to ASC while you're in there) Then, record a log file.  You should see those 0x7E8 packets in your log, but I don't think they contain any useful information that you could decode.

closed 2022-10-06 by jake-b