tesLAXApp/tesLAX #20: Canserver streaming data, but all gauges dead since last update

orcinus, 2021-12-02:
Since last TesLax update, all the gauges / dashboards have gone dead.
The only signals showing up are a couple of 12V battery ones. Everything else displays N/A.

Yes, CANserver and TesLax have been configured correctly and data is streaming.
Yes, i have verified the signals are showing up in CANserver itself.
And yes, i have reset the presets, and even reinstalled TesLax.

Any ideas?

jake-b, 2021-12-02:
Very odd.  The last update was a relatively minor one that only tweaked a few fonts and things to fix display issues on iOS 15.

Care to stop by the CANServer discord and see if we can troubleshoot?  https://discord.gg/9pP8Jakn

orcinus, 2021-12-02:
Will do.
I'm not sure if it was the last version or the one before that, to be honest - i haven't updated the app in a while.