tesLAXApp/tesLAX #26: Teslax cannot connect

kobusm, 2022-07-31:
Hi, I would really appreciate any support you can provide. I have a refresh Model S which Teslax cannot connect to.

What I  tried so far
1. Two different Tesla to OBD adapter cables. Latest from Maxwell Auto Tech that connects vehicle and chassis Can busses
2. I have tried the OBDLINK MX+ and 2 different OBDLINK CX devices. Confirmed MX+ works in Model 3 
3. I have tried Models S and Model 3 presets
4. Confirmed Can signals are present on connector with oscilloscope 

Always tries to connect, blinks blue and red LEDs and then says it fails to connect.

Please help.

Regards Kobus

jake-b, 2022-07-31:
Hi. trying to get you more info.  Sadly I don’t have a Model S refresh myself to test on. But here’s what I think is going on:

1.  Tesla changed the pinout of the connector.  The busses are swapped around. 
2. The community contacted Maxwell about updating the cable but they went non-responsive. 
3. When you use the old cable, the BFT bus (which contains no useful information) is routed to the standard OBD2 pins.  Hence why you see CAN.  There is CAN there but not a useful bus.

If I can get more info, I will post here. 

(Side note, if you are the type to poke at things with an oscilloscope, I’d check out JWardell’s CANserver. It’s a custom 2-bus interface that can do much more than the OBDLink hardware. Also the Discord community has more discussions of signals and busses.

kobusm, 2022-07-31:
Jake,

Thanks for the quick reply. Attached is what I  believe the pinout of the
20 pin connector is. CAN bus 3 powertrain/drive train/ vehicle is the one
you are monitoring. Is that correct? I  will be happy to scope the signals
if you can tell me how I  can confirm.

Here is a link to a video of what happens
https://share.icloud.com/photos/0bceoHEV33MrvNNUZm55N40Xg

Regards Kobus

jake-b, 2022-07-31:
I believe this is the pinout:
[image]

jake-b, 2022-08-02:
Not that you’re looking to but another cable, but I was just informed that EVOffee has one: https://evoffer.com/product/model-s-x-can-diagnostic-cable/

it says it works with tesLAX but I can’t verify that as I do not have this cable (or a model S)

kobusm, 2022-08-09:
> Jake,
>
> I am going to try to modify one of the cables I already bought if if ygst
> doesn't work buy this one. I will let you know
>
> Regards Kobus

Jake,

I modified the cable and it finally worked! Thanks for your help. One final
request. I am using the Plaid profile but I have the Model S LR so some
things like the rear motor power is not correctly displayed. Any idea when
there will be a profile for the S LR?

Regards Kobus

jake-b, 2022-08-12:
Glad you got it working.  I'd really like to update the profiles but the community hasn't discovered (or at least no one has published) the signals for the latest versions of the Tesla OS.  tesLAX is meant to be pretty agnostic (come with your own DBCs/signals) and I try to include somewhat up-to-date signals, but I do not have the resources to reverse engineer the car's OS and maintain the signal list.

kobusm, 2022-10-11:
Jake,

I am going to try to modify one of the cables I already bought if if ygst
doesn't work buy this one. I will let you know

Regards Kobus

Bobselby1, 2023-08-08:
Has anyone verified if the evoffer cable works with the refreshed s/x? Also if not can you verify what pin setup worked for you kobusm?