tesLAXApp/tesLAX #28: Front stator temp in Celsius shows 215

KoderGuy, 2022-09-10:
The temperature value for front stator in Celsius always shows 215c.  Occasionally it flashes the correct value for an instant but returns to 215 almost immediately.  I double checked in scan my Tesla and the temperature is actually 34c.  This just started happening.

Signal: tempStator376
Version: 2.0.3

Let me know what I can provide to help troubleshoot this.

Thanks
Richard Chong

jake-b, 2022-09-10:
This is an example of what happens when Tesla changes the data stream.  tesLAX is meant to be configurable so people can update the signals as they change.  I do not maintain the DBCs or signal decoding rules.  We rely on the community to do that.  I simply provide a super configurable app, and leave it up to the community to chase Tesla as they make changes

that said, there are a couple of profiles I posted in the tesLAX discord that will be part of the next release.  I believe this change is accounted for in those profiles.  If you’re not on discord I can post them the GitHub.

jake-b, 2022-09-10:
I have uploaded the TLP files into this git repository.  If you are an 'extra strength' user, you should be able to import them into your presets list.  You can also find me on Discord for additional help.

closed 2022-10-06 by jake-b