tesLAXApp/tesLAX #33: big-endian signal issue

bluntd42, 2023-07-29:
Hi Jake,

I am using App version 2.4.1 (171) with Extra Strength v2 to help me reverse engineer the CAN bus code in my GM car.

I have an issue where any tes-lax Gauges linked to big-endian signals (denoted @0 in a DBC file) that include the last byte of a CAN bus message do not display any values, or worse, cause the whole app to crash.

While investigating this, I discovered that the bits highlighted (with blue boxes) in Binary Signals for any big-endian signals are not displayed correctly. However, curiously, the Gauge values appear to be sourced from the correct bits. An example is shown below.

DBC file message with big-endian signals (created in SavvyCAN v199.1):
BO_ 405 x195: 8 Vector__XXX
   SG_ x195_0 : 4|1@0+ (1,0) [0|1] "" Vector__XXX
   SG_ x195_4 : 39|8@0+ (0.5,0) [0|128] "kPa" Vector__XXX
   SG_ x195_5 : 47|8@0+ (0.125,0) [0|32] "" Vector__XXX
   SG_ x195_67 : 55|16@0+ (0.0078125,0) [0|512] "" Vector__XXX

Binary Signals for the above CAN message:
 
[image] [image]
 
Using the convention that the initial bit for a big-endian signal is the most significant bit, and that the bits are shown in the following order:
 7  6  5  4  3  2  1  0
15 14 13 12 11 10  9  8
23 22 21 20 19 18 17 16
31 30 29 28 27 26 25 24
39 38 37 36 35 34 33 32
47 46 45 44 43 42 41 40
55 54 53 52 51 50 49 48
63 62 61 60 59 58 57 56

SG_ x195_0: shows bit 3 instead of bit 4
SG_ x195_4: shows bits 32 and 41-47 instead of bits 32-39.
SG_ x195_5: shows bits 40 and 49-55 instead of bits 40-47.

If I try to show a Binary Message for SG_ x195_67, the whole app crashes. I suspect this is because the bit pointers in this case extend beyond the end of the message into unallocated memory. I also suspect that this is why a Gauge linked to SG_ x195_67 does not show any values, as shown below. Also as shown below, despite the incorrect bits for SG_ x195_4 being highlighted, the gauge value is correct:

i.e. Row 5 x 0.5 = (01010111)binary x 0.5 = 87 x 0.5 = 43.5 kPa
The highlighted bits for this signal would actually give (10010101) binary x 0.5 = 149 x 0.5 = 74.5 kPa
 
[image] [image]

The little-endian signals (denoted @1 in a DBC file) created in SavvyCAN and imported into tes-lax work better, except for the single-bit SG_ x2C3_7_idle signal example below which is still off by one-bit (it highlights bit 59 instead of bit 60).

DBC file message with big-endian signals (created in SavvyCAN v199.1):
BO_ 707 x2C3_MAP: 8 Vector__XXX
   SG_ x2C3_01 : 0|16@1+ (0.0078125,0) [0|512] "" Vector__XXX
   SG_ x2C3_6_MAP : 48|8@1+ (0.5,0) [0|128] "kPa" Vector__XXX
   SG_ x2C3_7_idle : 60|1@1+ (1,0) [0|1] "flag" Vector__XXX

[image] [image]

Would you please investigate and let me know if I am on the right track? I am hoping that this will be a relatively easy fix. 

Regards, David

jake-b, 2023-07-29:
Thanks for a great bug report.  This is likely a bug.  The decode logic is some of my least favorite in the app and I have no doubt there could be a bug.

It sounds like maybe 2 bugs?  One in the matrix visualization as to what gets highlighted. And another in the decoding of the value that gets fed to a normal gauge.

If it’s not too much trouble an exported preset file and a short log would help, especially if you have one that reproduces the crash. 

You can find me on discord too if you’d like to talk more or send the file that way.  

https://discord.gg/K36KVw54

Thanks for a super awesome bug report.

bluntd42, 2023-07-29:
I exported my "Other CANBus Device.tlp" preset file, but I had to zip it in order to drag and drop it here. I get an unsupported file error if I try to drop the tpl file directly. I will try to drop it in discord as well.
[Other CANBus Device.zip](https://github.com/tesLAXApp/tesLAX/files/12208256/Other.CANBus.Device.zip)

The tesLaAX app crash that I mentioned happens as soon as I create a Binary Signal linked to SG_ x195_67 and save it. It crashes straight after I press save. I then can't restart the app - it blinks on then shuts down straight away - so there's no chance of generating a log file with that Binary Signal in place. To resurrect the app, I have to go into the tesLAX settings and select "discard database and restore default". I will try to generate a log file with the non-functioning Gauge linked to SG_ x195_67 later today.

I am curious as to whether only little-endian signals are used on Teslas.

bluntd42, 2023-07-30:
[2023-07-30 01.11.46 pm.asc.zip](https://github.com/tesLAXApp/tesLAX/files/12208547/2023-07-30.01.11.46.pm.asc.zip)
Here is 60 seconds of logged data in ASC format for the 0x195 and 0x2C3 messages.