tesLAXApp/tesLAX #34: How to connect?

Bobselby1, 2023-08-08:
Looking all over the place and for the life of me I cannot find any instructions or options to connect the app to my MX+ OBD reader. What am I missing?

jake-b, 2023-08-08:
Please see the help document in the app. (Tap upper left menu > help) and find “connecting to accessories”

The MX+ needs to be BOTH paired and connected in the Apple iOS settings app.  It should then appear in the app under “Accessories” > “Bluetooth”

M481248, 2023-10-23:
Having similar problems, I have tried both a OBDLinkMX+ and S3XY commander (separately).  I paired MX+ via Bluetooth and S3XY via Wifi.  On TesLAX, I can see the MX+ in accessories in the TesLAX app, but it never connects.  See Photo below. For S3XY, i have tried Panda and Wifi with same result.  Any suggestions?

[image]

jake-b, 2023-10-23:
Hmm.  That's two separate parts of the app that aren't working for you.  Are you getting data from other apps?  Wondering if it's a wiring issue.  What wiring harness are you using?

Wifi wise, you need to make sure you've allowed local network access, and I believe commander uses Panda, not ELM327-over-Wifi

Pop on discord to discuss, if you like: https://discord.gg/u2DkgvXG

Bobselby1, 2023-10-23:
Yes, the problem was the OBD2 adapter harness itself. I tried 3, the first two would not work. Ultimately got the one from EVOffer and it worked perfectly. Apparently the pin outs on some of the other ones just aren’t correct.

[WIN_20231023_12_52_38_Pro]<https://user-images.githubusercontent.com/148797914/277413103-1b266caa-4783-43c1-acd1-c421201e4879.jpg>

M481248, 2023-10-23:
My harness is good.  Scan my Tesla (android) works fine and it works with my iphone (iOS 16).  I believe it is an issue now with IpadOS17 running on the iPad.  Any idea if iPadOS17 is supported with Tes LAX?

M481248, 2023-10-24:
Jake,Can confirm my issue is with the iPadOS 17 as I borrowed an iPad running is iPadOS 15 and was able to collect the data as expected. Have you heard of other issues with those running OS17 (assuming iOS 17 would give similar issues as iPadOS 17). Thanks

jake-b, 2023-10-24:
Hi.  This is the first report of an issue specific to iPadOS.  I will check it out on my iPad mini and see if I can reproduce/debug.  Not sure why there would be a difference in behavior

stromi-tesla-model3, 2023-10-26:
Can report there are no problems with Tes•LAX 2.4.2 (174) on iPadOS 17.0.3 on iPad mini (5th Gen)

M481248, 2023-10-26:
Good to know Stomi - Thanks!   at least I now know its the device :)  I will play around with it.

stromi-tesla-model3, 2023-11-04:
> Sometimes MX+ will switch devices.

Don't forget to apply any firmware updates to the MX+ via the OBDLink app. OBDLink may seem broken at startup, as it doesnt "connect to car", but with OBDLink and MX+ connected, you can check for and apply firmware updates.

SPlatten, 2024-02-27:
What do you need to connect the app to a Tesla?  I have various ODBCII connectors and I've been investigating the CAN-BUS with SavvyCAN and connecting to my Tesla CAN-BUS via the centre console, but how would I connect Tes-LAX to the car?