tesLAXApp/tesLAX #40: Issue Connecting Tesla Model 3 (2022) Using Panda Protocol with S3XY Commander

moroz-alexander, 2024-09-16:
Hello,

I’m trying to connect to my Tesla Model 3 (2022) using the S3XY Commander, which utilizes the Panda protocol. When I use the ScanMyTesla app, the connection works flawlessly. However, I’m having trouble with your app. Could you please provide guidance on the settings I should use to make the connection work?

Thank you in advance!

jake-b, 2024-09-17:
HI!  I don't have a commander, but there are many people that report it works.  The only thing I can think of is that the commander uses a different IP address.  The default `192.168.0.10` IP address is for an actual Comma.ia Panda White.  This is kind of an unusual address for something like this.  I think most things (and the Commander?) user `192.168.4.1`

moroz-alexander, 2024-09-20:
I was able to connect to the S3XY! commander using the following settings:

	•	Host IP: 192.168.4.1
	•	UDP Port: 1338
	•	Passive Mode: Disabled
	•	Auto-connect to Panda: Enabled

[image](https://github.com/user-attachments/assets/b71f3fdf-b05f-4eaa-ae58-afaceab1bc13)

moroz-alexander, 2024-09-20:
The issues didn’t stop there. As you can see, the connection was successful, and everything seems fine. The message queue is being processed at a good speed—currently 380 messages per second, which indicates a stable connection. However, despite the data exchange, no information is being displayed on the dashboards. All values in the UI remain "N/A," and parameters such as battery, motor, or speed are not being shown. It seems like the data isn’t being retrieved correctly, even though the connection has been established.

The current software version of the car:  
**v12 (2024.32.3 2e59c0a47060)**  
Navigation data: **EU-2024.20-15072-6ad3777f90**.

[image]

[image]

jake-b, 2024-09-20:
OK this sounds like a bus misconfiguration.  I'm not super familar with the Commander, but I believe its a two-bus device.

tesLAX's presets expect the Vehicle Bus to be on the first bus (bus 0) and the Chassis bus to be on the second bus (bus 1). This be cause 1-bus devices (which only have bus 0) are usually connected to the vehicle bus, so it maintains compatibility.

There are two ways to configure this --
1. on the device, if there's a setting for which bus is assigned which number.  This is my preferred method, but I understand if people use other apps or have specific setups where they want Chassis to be on 0.
2. you can switch it in the preset--  in the preset list, tap the (i) button next to the preset you want to edit.  You'll see the "can bus numbering" section -- tap edit and then "Bus Channels" - you can reverse the bus channels.

I don't prefer #2 because you have to do it with every preset, and if you get new presets, you have to do it with each new preset.  But either method should work.

moroz-alexander, 2024-09-21:
Thank you for the information. It might be useful for others as well. I selected Channel 2 and set it to "Deliver to All." After that, everything started working for me.

[image]

jake-b, 2024-09-23:
hey this turns out ot be a bug in a recent Commander firmware update.  They will be eventually sending out a new firmware with fixes for Panda compatibility.   You can see my note here: 

https://github.com/tesLAXApp/tesLAX/issues/41#issuecomment-2367986528