tesLAXApp/tesLAX #41: Tes.lax only reporting batery capacity and a few more.

alvesjc, 2024-09-18:
Hi,

for some reason almost all the values stopped working.

in Appstore I see that there is no updates for more than a year.

Did the PIDs change meanwhile and I’ve to manually update them somehow?

I’m connecting with Sexy commander

thanks

jake-b, 2024-09-20:
Are you getting good data flow?  Does the app shows packets coming to the app?

I have some updates planned, but the community isn't generating new signal decoding information like it used to.  As tesla changes the signals, we have a hard time keeping up.  The last useful set of changes we had was about a year ago, yeah.

alvesjc, 2024-09-20:
Hi @jake-b,

Yes I've packets coming in, and some values are updated.
It started with some wrong values like the 12V batery and some NA, but now it's almost all NA.
I think this got worse with 2024.26 update.

jake-b, 2024-09-20:
OK this sounds like an issue with reversed busses. Take a look at this response I just made: 

https://github.com/tesLAXApp/tesLAX/issues/40#issuecomment-2363453620

alvesjc, 2024-09-20:
I'll test and let you know

alvesjc, 2024-09-20:
So, I reversed the buses has you said, it started to work, but there is still issues.
The SuC charging current still says N/A, and used to show 0 when not charging.
The 12V battery is still messed up also, the 12V AH shows(15.5) what I believe is actually the voltage since my car uses already Lithium 12V battery, and voltage shows 8.6, that might this actually be the battery capacity in AH.

I also now have a warning message about the multibus devices that is constantly showing up.

[image]
[image]

jake-b, 2024-09-23:
So I reached out to Enhanced. A recent update to the Commander is the issue.  The reversed bus thing you tried was an interesting data point.  They are working on a new release that fixes Panda support. My guess is you'll have to put the buses back the way they were once the new firmware is available.

The bus warning is dismissable.  It is intended to warn 1-bus users when there are signals on the page that require a bus other than bus 0.  (which when you flip the buses, you do).  You should be able to dismiss and they won't come back. 

The battery thing is an example of tesla changing the signals.  The community used to be much more free flowing with signal data.  I'll see if I can dig up some updates for the signal definitions.  Though I make it clear that I am not in the business of selling signal data.  TesLAX is a configurable framework that allows the user to configure the signal data.  I just provide some initial presets to get people start.